mpromonet/v4l2tools #2: Raspberry Pi Camera Module

chrisf4lc0n, 2015-12-20:
When issuing command:
 sudo v4l2copy /dev/video0 /dev/video1

I keep getting:
VIDIOC_REQBUFS: Inappropriate ioctl for device error

mpromonet, 2016-01-04:
Hi Krzysztof,
It seems /dev/video1 doesnot support mmap access ?
Could you post the result of : 

    v4l2-ctl --all -d /dev/video1
In order to use write api for output instead of mmap, you can try to add the `-w` option (I add a new option in commit 8e7c1bf6a152bdf6ab0edbe9ecb5f4fb30e30542).

Best Regards,
Michel.

chrisf4lc0n, 2016-01-04:
Driver Info (not using libv4l2):
	Driver name   : bm2835 mmal
	Card type     : mmal service 16.1
	Bus info      : platform:bcm2835-v4l2
	Driver version: 4.1.15
	Capabilities  : 0x85200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x05200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 0: ok)
Format Video Capture:
	Width/Height  : 640/480
	Pixel Format  : 'H264'
	Field         : None
	Bytes per Line: 0
	Size Image    : 307200
	Colorspace    : Broadcast NTSC/PAL (SMPTE170M/ITU601)
	Flags         : 
Format Video Overlay:
	Left/Top    : 150/50
	Width/Height: 1024/768
	Field       : None
	Chroma Key  : 0x00000000
	Global Alpha: 0x00
	Clip Count  : 0
	Clip Bitmap : No
Framebuffer Format:
	Capability    : 
	Flags         : Overlay Matches Capture/Output Size
	Width         : 640
	Height        : 480
	Pixel Format  : 'YU12'
	Bytes per Line: 640
	Size image    : 460800
	Colorspace    : Broadcast NTSC/PAL (SMPTE170M/ITU601)
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 20.000 (20/1)
	Read buffers     : 1

User Controls

                     brightness (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                       contrast (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                     saturation (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                    red_balance (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip (bool)   : default=0 value=0
                  vertical_flip (bool)   : default=0 value=0
           power_line_frequency (menu)   : min=0 max=3 default=1 value=1
                      sharpness (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                  color_effects (menu)   : min=0 max=15 default=0 value=0
                         rotate (int)    : min=0 max=360 step=90 default=0 value=0
             color_effects_cbcr (int)    : min=0 max=65535 step=1 default=32896 value=32896

Codec Controls

             video_bitrate_mode (menu)   : min=0 max=1 default=0 value=0 flags=update
                  video_bitrate (int)    : min=25000 max=25000000 step=25000 default=10000000 value=10000000
         repeat_sequence_header (bool)   : default=0 value=0
            h264_i_frame_period (int)    : min=0 max=2147483647 step=1 default=60 value=60
                     h264_level (menu)   : min=0 max=11 default=11 value=11
                   h264_profile (menu)   : min=0 max=4 default=4 value=4

Camera Controls

                  auto_exposure (menu)   : min=0 max=3 default=0 value=0
         exposure_time_absolute (int)    : min=1 max=10000 step=1 default=1000 value=1000
     exposure_dynamic_framerate (bool)   : default=0 value=0
             auto_exposure_bias (intmenu): min=0 max=24 default=12 value=12
      white_balance_auto_preset (menu)   : min=0 max=9 default=1 value=1
            image_stabilization (bool)   : default=0 value=0
                iso_sensitivity (intmenu): min=0 max=4 default=0 value=0
         exposure_metering_mode (menu)   : min=0 max=2 default=0 value=0
                     scene_mode (menu)   : min=0 max=13 default=0 value=0

JPEG Compression Controls

            compression_quality (int)    : min=1 max=100 step=1 default=30 value=30
                     brightness (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                       contrast (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                     saturation (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                    red_balance (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip (bool)   : default=0 value=0
                  vertical_flip (bool)   : default=0 value=0
           power_line_frequency (menu)   : min=0 max=3 default=1 value=1
                      sharpness (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                  color_effects (menu)   : min=0 max=15 default=0 value=0
                         rotate (int)    : min=0 max=360 step=90 default=0 value=0
             color_effects_cbcr (int)    : min=0 max=65535 step=1 default=32896 value=32896

chrisf4lc0n, 2016-01-04:
Hi again,
It seems like -w option does something as I am not receiving any errors now, but still unable to get connected to the /dev/video1
All the best Krzysztof.

mpromonet, 2016-01-04:
Hi,
What is your /dev/video1 ? a real output device ? a virtual device like v4l2loopback ?
If v4l2-ctl is not able to query it, it is probably not a V4L2 device.
Regards,
Michel.

chrisf4lc0n, 2016-01-04:
I think I got to the bottom of it... It seems to be a problem with v4l2loopback.
Thanks for everything.

closed 2016-01-04 by chrisf4lc0n

Apteryks, 2023-05-24:
@chrisf4lc0n would you mind sharing what was the problem with the v4l2loopback module?  I'm facing the same problem.

Apteryks, 2023-05-24:
OK, after `apt purge v4l2loopback-dkms` and `apt install 4l2loopback-dkms` followed by a reboot, it works normally again.  Phew.