mpromonet/v4l2tools #3: v4l2compress_omx

idreamerhx, 2015-12-26:
hi~

I use v4l2compress_omx on archlinux on raspberry with a use camera.

The camera:
[root@betta v4l2tools]# v4l2-ctl --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : USB2.0 YW500 Camera
	Bus info      : usb-20980000.usb-1.3.6
	Driver version: 4.1.15
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height  : 640/480
	Pixel Format  : 'YUYV'
	Field         : None
	Bytes per Line: 1280
	Size Image    : 614400
	Colorspace    : SRGB
	Flags         : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness (int)    : min=0 max=255 step=1 default=-8193 value=110
                       contrast (int)    : min=0 max=255 step=1 default=57343 value=40
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=56
                            hue (int)    : min=-45 max=45 step=1 default=-8193 value=-10
 white_balance_temperature_auto (bool)   : default=1 value=1
                          gamma (int)    : min=102 max=200 step=1 default=57343 value=120
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature (int)    : min=2800 max=6500 step=1 default=57343 value=4500 flags=inactive
                      sharpness (int)    : min=0 max=255 step=1 default=57343 value=64
         backlight_compensation (int)    : min=0 max=4 step=1 default=57343 value=1
                  exposure_auto (menu)   : min=0 max=3 default=0 value=3
              exposure_absolute (int)    : min=3 max=2047 step=1 default=166 value=166 flags=inactive
                     brightness (int)    : min=0 max=255 step=1 default=-8193 value=110
                       contrast (int)    : min=0 max=255 step=1 default=57343 value=40
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=56
                            hue (int)    : min=-45 max=45 step=1 default=-8193 value=-10
 white_balance_temperature_auto (bool)   : default=1 value=1
                          gamma (int)    : min=102 max=200 step=1 default=57343 value=120
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature (int)    : min=2800 max=6500 step=1 default=57343 value=4500 flags=inactive
                      sharpness (int)    : min=0 max=255 step=1 default=57343 value=64
         backlight_compensation (int)    : min=0 max=4 step=1 default=57343 value=1

The probable bug maybe:

2015-12-26 12:36:41,977 [NOTICE] - src/V4l2MmapDevice.cpp:55
	Device /dev/video0 nb buffer:10
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
Segmentation fault (core dumped)

I use gdb compared v4l2compress_h264 (first, works but very slow) and v4l2compress_omx(second):

(gdb) p m_buffer
$1 = {{start = 0xb64ce000, length = 614400}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0,
    length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}}
(gdb) p m_fd
$2 = 3
(gdb) p buf
$3 = {index = 0, type = 1, bytesused = 0, flags = 73728, field = 0, timestamp = {tv_sec = 0, tv_usec = 0}, timecode = {type = 0, flags = 0, frames = 0 '\000', seconds = 0 '\000', minutes = 0 '\000', hours = 0 '\000',
    userbits = "\000\000\000"}, sequence = 0, memory = 1, m = {offset = 0, userptr = 0, planes = 0x0, fd = 0}, length = 614400, reserved2 = 0, reserved = 0}

(gdb) p m_buffer
$2 = {{start = 0xffffffff, length = 462848}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}, {start = 0x0,
    length = 0}, {start = 0x0, length = 0}, {start = 0x0, length = 0}}
$1 = {index = 0, type = 1, bytesused = 0, flags = 73728, field = 0, timestamp = {tv_sec = 0, tv_usec = 0}, timecode = {type = 0, flags = 0, frames = 0 '\000', seconds = 0 '\000', minutes = 0 '\000', hours = 0 '\000',
    userbits = "\000\000\000"}, sequence = 0, memory = 1, m = {offset = 2882400000, userptr = 2882400000, planes = 0xabcdef00, fd = -1412567296}, length = 462848, reserved2 = 0, reserved = 0}

mpromonet, 2016-01-04:
Hi idreamerhx,
I am not sure to follow you, especially on which line occurs the core dump ?
It seems the mmap failed because in your gdb investigation `$2 = {{start = 0xffffffff, length = 462848}`, start=(void*)-1 indicate mmap failed and the length seems too short (it should be the same as with x264, I means 614400). Then perhaps there is no more mmap memory available.
By the way I add a check that the capture start sucessfully in commit ff9350b3ff72a8affe3978f38b090c02ad63f31a. 
With this modification, I guess it will stop reporting capture could not start ?
Best Regards,
Michel.

idreamerhx, 2016-01-08:
Thanks for your reply, I just pull the code and try.

idreamerhx, 2016-01-13:
Hi~

I found my camera support only V4L2_PIX_FMT_YUYV, so I change V4L2_PIX_FMT_YUV420 to it. In addition, I add a call to ConvertToI420 like it is in v4l2compress_h264.cpp to v4l2compress_omx.cpp.

There is no core dump for the latest code. But, I found ilclient_get_input_buffer returns a buffer which may not have a correct length. I break in line 156: 

buf = ilclient_get_input_buffer(video_encode, 200, 0);

Please see the gdb print:

(gdb) p *buf
$9 = {nSize = 72, nVersion = {s = {nVersionMajor = 1 '\001', nVersionMinor = 1 '\001', nRevision = 2 '\002', nStep = 0 '\000'}, nVersion = 131329},
  pBuffer = 0xb298f020 "", nAllocLen = 460800, nFilledLen = 0, nOffset = 0, pAppPrivate = 0x0, pPlatformPrivate = 0x0, pInputPortPrivate = 0x1f4ecdc0,
  pOutputPortPrivate = 0x58e10, hMarkTargetComponent = 0x0, pMarkData = 0x0, nTickCount = 0, nTimeStamp = 0, nFlags = 0, nOutputPortIndex = 200,
  nInputPortIndex = 0}

(gdb) p *video_encode
$10 = {comp = 0x58d30, flags = (ILCLIENT_ENABLE_INPUT_BUFFERS | ILCLIENT_ENABLE_OUTPUT_BUFFERS | ILCLIENT_DISABLE_ALL_PORTS), sema = {
    __size = "\002", '\000' <repeats 14 times>, __align = 2}, event = {events = 92, lock = {__data = {__lock = 0, __count = 0, __owner = 0, __kind = 0,
        __nusers = 0, {__spins = 0, __list = {__next = 0x0}}}, __size = '\000' <repeats 23 times>, __align = 0}, waiters = {head = 0x0, tail = 0x0}},
  related = 0x0, out_list = 0x68eb0, in_list = 0x0, name = "cl:video_encode:2", '\000' <repeats 14 times>,
  bufname = "cl:video_encode:2 buffer\000\000\000\000\000\000\000", error_mask = 0, private = 0, list = 0x4f620, client = 0x4ee50}

(gdb) p *videoCapture
$11 = {<V4l2Device> = {_vptr.V4l2Device = 0x34654 <vtable for V4l2MmapCapture+84>, m_params = {m_devName = "/dev/video0", m_format = 1448695129,
      m_width = 640, m_height = 480, m_fps = 10, m_verbose = 0}, m_fd = 3}, _vptr.V4l2Capture = 0x34630 <vtable for V4l2MmapCapture+48>,
  m_bufferSize = 614400, m_format = 1448695129, m_width = 640, m_height = 480}

this problem maybe cased by ilclient. but I'm not sure it due to my compile.

mpromonet, 2016-01-13:
Reading quickly, I guess the problem comes because `I420` is not `YUV420`, and the size are different.

idreamerhx, 2016-01-27:
Thank you very much. These days I search and try how to solve the problem off and on. If I have any progress, I will share it there.

IMSancho, 2016-02-01:
I was having similar issues trying to get a YUYV camera to work, and was able to sort it out by using the converttoI420 code from the H264 example, and doing the conversion first before inserting into the ilclient buffer. Code for the input section now looks like this:
```
buf = ilclient_get_input_buffer(video_encode, 200, 0);
	if (buf != NULL)
	{
			char buffer[videoCapture->getBufferSize()];
			int rsizez = videoCapture->read(buffer, sizeof(buffer));
			int upos = width * height;
			int vpos = upos + upos / 4;

			ConvertToI420((const uint8*)buffer, rsizez,
					buf->pBuffer, width,
					&buf->pBuffer[upos], width/2,
					&buf->pBuffer[vpos], width/2,
					0, 0,
					width, height,
					width, height,
					libyuv::kRotate0, libyuv::FOURCC_YUY2);

			/* fill it */
			int rsize = vpos * 2 - upos;
			LOG(DEBUG) << "read size:" << rsize << " buffer size:" << buf->nAllocLen;

			buf->nFilledLen = rsize;

			if (OMX_EmptyThisBuffer(ILC_GET_HANDLE(video_encode), buf) != OMX_ErrorNone)
			{
					fprintf(stderr, "Error emptying buffer!\n");
			}
	}
```
I420 and YV12 are the same except for having the chroma order reversed, so size is the same, the problem with trying to insert a YUYV frame into a buffer for a 420 frame is that while they both have width * height Y values, the 420 only has a U and V for every 4 pixels, not every 2 like a YUYV, hence the buffer truncation and distorted output. With the above changes I am streaming at 640x480x25fps from an old webcam with about 35% cpu utilisation from v4l2compress_omx on a pi B+

idreamerhx, 2016-02-24:
For these days, I occasionally tried some methods. On my new rpi B 2 (1G memory), I use libyuv convert yuyv to i420, it works. thanks to the neon acceleration of libyuv the cpu usage is about 10% for 640x480 25fps. The video seems not so good,  i don't know why.

Another thing is i tried opengl shader to convert yuyv to rgb888. For these days i finished this for a single frame. and i'll try to combine it to v4l2tools.

I'll post here if i finished.

closed 2017-09-04 by mpromonet