mpromonet/v4l2tools #4: v4l2copy not compatible with ffmpeg

idreamerhx, 2016-08-24:
Hi, 

I use ./v4l2copy  /dev/video0  /dev/video10 to copy from raw camera (video0) to v4l2loopback (video10).

Then, ffmpeg -y -f v4l2 -i /dev/video10 -r 1 -vframes 10  output%3d.jpg

but ffmpeg only output 2 frames, dropping all others. For video0, there is no this issus.

I got this problem when I try to do:

ffmpeg -f v4l2 -video_size 640x480 -ss 0 -vsync 1  -copyts -i /dev/video10 out1.avi

2147483646 frame duplication too large, skipping0:00:00.00 bitrate=N/A dup=0 drop=26 speed=   0x    
    Last message repeated 3 times
2147483646 frame duplication too large, skipping0:00:00.00 bitrate=N/A dup=0 drop=30 speed=   0x    
    Last message repeated 3 times
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
[rawvideo @ 0x5631e510] PACKET SIZE: 921600, STRIDE: 1920
cur_dts is invalid (this is harmless if it occurs once at the start per stream)op=34 speed=   0x    
[rawvideo @ 0x5631e510] PACKET SIZE: 921600, STRIDE: 1920
cur_dts is invalid (this is harmless if it occurs once at the start per stream)

Currently I think it may be something wrong with v4l2copy.

mpromonet, 2016-08-24:
Hi idreamerhx,
Did you tried to add `-r -w` in order to use read/write instead of mmap interface ?
I remember a bug in v4l2loopback using compress format with mmap interface (but this is fixed since a while).
Best Regards,
Michel.

idreamerhx, 2016-08-25:
Thanks for your quickly reply and help. -w is OK for me to solve the problem.

It seams my camera not support mmap.

[root@betta v4l2tools]# ./v4l2copy_gpuplay_rgb888 -r -w /dev/video0  /dev/video10
2016-08-25 10:26:50,873 [NOTICE] - src/V4l2Device.cpp:131
	driver:uvcvideo 84200001
2016-08-25 10:26:50,873 [NOTICE] - src/V4l2Device.cpp:141
	/dev/video0 support streaming
2016-08-25 10:26:50,874 [ERROR ] - src/V4l2Device.cpp:146
	Mandatory capability not available for device:/dev/video0

video0:
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities

Thank you again for your handy tools. It realy important to my fishbowl supervisory.

closed 2016-08-25 by idreamerhx

idreamerhx, 2016-08-25:
So, is this a v4l2loopback bug or issue?

mpromonet, 2016-08-28:
This issue is fixed since this [commit](https://github.com/umlaeute/v4l2loopback/commit/f8cccb964727fdb697f6190986d76d750781ef81)