mpromonet/v4l2tools #5: Miner update/edit

bdheeman, 2017-02-21:
Please have a look at files attached herewith ;-)
[edit.zip](https://github.com/mpromonet/v4l2tools/files/789126/edit.zip)

mpromonet, 2017-02-21:
Hi bdheeman,
Thanks for the submission, however it is better to describe what you would like to modify and submit as a PullRequest.
Adding a PREFIX mechanism could be interesting, but the DESTDIR seems useless.
Regards,
Michel.

bdheeman, 2017-02-21:
I think, DESTDIR is a must for binary packaging on almost all systems.

These were quite a minor edit/update and, or changes; I added a few missing entries to .gitignore as well, to avoid unwanted addition of binary blobs and, or commits.

mpromonet, 2017-02-22:
Why do you think DESTDIR is must, as you install in ${DESTDIR}${PREFIX}, you can simply add what you like in the PREFIX ?

bdheeman, 2017-02-22:
In most of all the other good software projects, people use DESTDIR and, or INSTALL_DIR directives to install software out of place e.g. in /tmp/package-whatever for packaging etc.

Whereas, PREFIX mostly is used for rpath utilities and linker etc., to set and, or use a specific or standard path for dynamically loadable libs/modules.

Although, this has nothing to do with your project specifically, but there's no harm following industry standards/conventions and learning or developing good coding/programming habits ;-)

mpromonet, 2017-02-22:
I saw so many standards/conventions, this project try to be as simple as it can.
Feel free to fork and follow all these good coding habits.

closed 2017-02-22 by mpromonet