mpromonet/v4l2tools #6: v4l2convert_YUV missing support for width and height parameters

roger-, 2017-03-15:
The command line help says it's supported but the code is missing. I can't get it to work in my application because the default size is crazy big:

    Cannot rescale input:640x480 output:1931488544x1445946226

mpromonet, 2017-03-15:
Hi Roger,
v4l2convert_YUV is trying to set the output (width,height) to the input (width,height) and this failed (because the output device doesnot support this resolution or an other reason).
Next this log trace show the requested resolution for output and the current resoplution of output, as it is different the program exit.
It might have some log that give information why the ioctl fails.
Best Regards,
Michel.

roger-, 2017-03-15:
Hi Michel,

Ah, I've actually misunderstood what this was for -- I thought the output was a virtual device. That said the help does mention width and height parameters which aren't actually supported. 

I'm actually trying to go from YUV-> h264+RTSP (using the OMX encoder on a Pi); can this be done with your tools?

Regards

mpromonet, 2017-03-16:
Hi Roger,

Yes the output could be a virtual device or a real one, it just needs to implement the V4L2 output interface.
If you like to rescale, you need to implement it, possibly with the livyuv scale function.

Using v4l2compress_omx, you can compress YUV420 (alias YU12) from one V4L2 device to an other one, if your YUV source doesnot support this YUV format, you could use v4l2convert_YUV to convert it.

If I remember well, the OMX encoder of the Pi support only the YU12 format.

Best Regards,
Michel.

roger-, 2017-03-19:
Thanks Michel, I was able to get it working but with one minor issue. After loading v4l2loopback I run:

```
v4l2convert_yuv /dev/video0 /dev/video1
v4l2compress_omx /dev/video1 /dev/video2
```

The last command gives me these messages:

```
2017-03-19 19:02:41,778 [NOTICE] - src/v4l2compress_omx.cpp:138
        Start Compressing /dev/video1 to /dev/video2
2017-03-19 19:02:41,780 [WARN  ] - src/V4l2MmapDevice.cpp:190
        Device /dev/video1 buffer truncated available:460800 needed:462848
2017-03-19 19:02:41,805 [WARN  ] - src/V4l2MmapDevice.cpp:190
        Device /dev/video1 buffer truncated available:460800 needed:462848
2017-03-19 19:02:41,871 [WARN  ] - src/V4l2MmapDevice.cpp:190
        Device /dev/video1 buffer truncated available:460800 needed:462848
```

I'm able to get everything working by adding the `-r` option to `v4l2compress_omx`, but I'm not sure why it's necessary.

closed 2017-09-04 by mpromonet