mpromonet/v4l2tools #9: v4l2convert_yuv problems with yuyv input

WiredLife, 2017-09-24:
because v4l2compress_omx doesnt take yuyv input i tried to convert it with v4l2convert_yuv to yu12 before
but the picture looks very strange
is there a solution to convert yuyv to h264 with omx?
v4l2compress_h264 is taking too much cpu on my raspberry so its getting hot

thanks!

mgoeller, 2019-07-08:
I have the same question. Want to convert a YUYV422 input from /dev/video0 hardware accelerated to h264 and output to a /dev/video1 (v4l2loopback device) with a Raspberry Pi.

mpromonet, 2019-09-05:
Hi WiredLife,
I modified v4l2compress_omx to support yuyv format (it convert to yuv420p if needed), it should work starting from https://github.com/mpromonet/v4l2tools/commit/270a144fee8a9d5ecb6c81efc82271f6c212f1ec

Best Regards,
Michel.

closed 2020-02-15 by mpromonet