mpromonet/v4l2tools #17: v4l2compress_h264 can't be found

gounthar, 2018-11-07:
Michel,

as stated in https://github.com/mpromonet/v4l2rtspserver/issues/114, I would like to try to stream the Raspberry Pi HDMI output, but in the sample I found, there is a reference to v4l2compress_h264.
I clone, compiled and installed this repo, but it looks like the v4l2compress_h264 is not compiled anymore.
The source file is there though...
Is there something I missed?

Thanks a lot for your work.

mpromonet, 2018-11-07:
Hi Bruno,

In order to build v4l2compress_h264 you need to install x264, the makefile compile tools depending of what is available on your system.
 
You can get it from https://circleci.com/gh/mpromonet/v4l2tools/30#artifacts/

Best Regards,

Michel.

gounthar, 2018-11-08:
Bingo!
```bash
sudo make install
with log4cpp
mkdir -p /usr/bin
install -D -m 0755 v4l2copy v4l2convert_yuv v4l2source_yuv v4l2dump v4l2grab_h264 v4l2display_h264 v4l2compress_omx v4l2compress_h264 v4l2compress_vpx v4l2compress_jpeg v4l2uncompress_jpeg /usr/bin

```

closed 2018-11-09 by mpromonet

robertlzj, 2022-07-01:
> the makefile compile tools depending of what is available on your system.

Hi, so hard for me to get `v4l2compress_omx`.
As I had thought it should be auto build, since I install all the [Dependencies](https://github.com/mpromonet/v4l2rtspserver#using-v4l2loopback), and use RPi 4 of course.
Finally, after I build [VideoCore userland](https://github.com/raspberrypi/userland), v4l2tools can make [Tools for Raspberry](https://github.com/mpromonet/v4l2tools#tools-for-raspberry).
I use Raspbian Lite (bullseye), seems which don't contain the pre-compiled GPU binary.

I searched through a lot, this post is the most related one, at least similar prompt "can't be found".
Related:
- [Raspberry Pi VideoCore userland build/install (/opt/vc) / Documentation / Dev1 Galaxy Forum](http://dev1galaxy.org/viewtopic.php?id=2967)
- [STICKY: All about accelerated video on the Raspberry Pi [thanks all contributors!] - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=317511)