mpromonet/v4l2tools #18: v4l2compress_h264 : VIDIOC_REQBUFS: Device or resource busy

dony71, 2018-12-29:
Run  "v4l2compress_h264 /dev/video0 /dev/video10 -vvv"
got error below
-----------------------------------------------------------------------------------------------------------
2018-12-28 16:50:34,577 [INFO  ] - v4l2wrapper/inc/logger.h:44
        level:DEBUG
2018-12-28 16:50:34,578 [NOTICE] - src/V4l2Device.cpp:117
        driver:ov534 capabilities:85200001 mandatory:4000001
2018-12-28 16:50:34,578 [NOTICE] - src/V4l2Device.cpp:120
        /dev/video0 support capture
2018-12-28 16:50:34,578 [NOTICE] - src/V4l2Device.cpp:122
        /dev/video0 support read/write
2018-12-28 16:50:34,578 [NOTICE] - src/V4l2Device.cpp:123
        /dev/video0 support streaming
2018-12-28 16:50:34,578 [NOTICE] - src/V4l2Device.cpp:147
        /dev/video0:YUYV size:640x480
2018-12-28 16:50:34,618 [NOTICE] - src/V4l2Device.cpp:229
        fps:1/25
2018-12-28 16:50:34,618 [NOTICE] - src/V4l2Device.cpp:230
        nbBuffer:2
VIDIOC_REQBUFS: Device or resource busy
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2Device.cpp:117
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video10 support output
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2Device.cpp:120
        /dev/video10 support capture
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2Device.cpp:122
        /dev/video10 support read/write
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2Device.cpp:123
        /dev/video10 support streaming
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2Device.cpp:147
        /dev/video10:H264 size:640x480
2018-12-28 16:50:34,619 [NOTICE] - src/V4l2MmapDevice.cpp:71
        Device /dev/video10 nb buffer:8
2018-12-28 16:50:34,619 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:0 size:0
mmap: Invalid argument
2018-12-28 16:50:34,619 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:1 size:0
mmap: Invalid argument
2018-12-28 16:50:34,620 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:2 size:0
mmap: Invalid argument

-----------------------------------------------------------------------------------------------------------
v4l2-ctl --all -d /dev/video10
Driver Info (not using libv4l2):
        Driver name   : v4l2 loopback
        Card type     : Dummy video device (0x0000)
        Bus info      : platform:v4l2loopback-000
        Driver version: 4.14.78
        Capabilities  : 0x85208003
                Video Capture
                Video Output
                Video Memory-to-Memory
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x85208003
                Video Capture
                Video Output
                Video Memory-to-Memory
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
Priority: 0
Video input : 0 (loopback: ok)
Video output: 0 (loopback in)
Format Video Output:
        Width/Height      : 640/480
        Pixel Format      : 'H264'
        Field             : None
        Bytes per Line    : 0
        Size Image        : 1228800
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             :
Streaming Parameters Video Capture:
        Frames per second: 30.000 (30/1)
        Read buffers     : 8
Streaming Parameters Video Output:
        Frames per second: 30.000 (30/1)
        Write buffers    : 8
User Controls
                    keep_format (bool)   : default=0 value=0
              sustain_framerate (bool)   : default=0 value=0
                        timeout (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io (bool)   : default=0 value=0

-----------------------------------------------------------------------------------------------------------
 v4l2-ctl --all -d /dev/video0
Driver Info (not using libv4l2):
        Driver name   : ov534
        Card type     : USB Camera-B4.09.24.1
        Bus info      : usb-1c1c000.usb-1
        Driver version: 4.14.78
        Capabilities  : 0x85200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x05200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (ov534: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'YUYV'
        Field             : None
        Bytes per Line    : 1280
        Size Image        : 614400
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             :
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 25.000 (25/1)
        Read buffers     : 2
User Controls
                     brightness (int)    : min=0 max=255 step=1 default=0 value=0 flags=slider
                       contrast (int)    : min=0 max=255 step=1 default=32 value=32 flags=slider
                     saturation (int)    : min=0 max=255 step=1 default=64 value=64 flags=slider
                            hue (int)    : min=-90 max=90 step=1 default=0 value=0 flags=slider
        white_balance_automatic (bool)   : default=1 value=1
                       exposure (int)    : min=0 max=255 step=1 default=120 value=253 flags=inactive, volatile
                 gain_automatic (bool)   : default=1 value=1 flags=update
                           gain (int)    : min=0 max=63 step=1 default=20 value=255 flags=inactive, volatile
                horizontal_flip (bool)   : default=0 value=0
                  vertical_flip (bool)   : default=0 value=0
           power_line_frequency (menu)   : min=0 max=1 default=0 value=0
                      sharpness (int)    : min=0 max=63 step=1 default=0 value=0 flags=slider
Camera Controls
                  auto_exposure (menu)   : min=0 max=1 default=0 value=0 flags=update

dony71, 2018-12-31:
v4l2compress_h264 -r /dev/video0 /dev/video10 -vvv
2018-12-31 10:30:07,717 [INFO  ] - v4l2wrapper/inc/logger.h:44
        level:DEBUG
2018-12-31 10:30:07,717 [NOTICE] - src/V4l2Device.cpp:117
        driver:ov534 capabilities:85200001 mandatory:1000001
2018-12-31 10:30:07,717 [NOTICE] - src/V4l2Device.cpp:120
        /dev/video0 support capture
2018-12-31 10:30:07,717 [NOTICE] - src/V4l2Device.cpp:122
        /dev/video0 support read/write
2018-12-31 10:30:07,718 [NOTICE] - src/V4l2Device.cpp:123
        /dev/video0 support streaming
2018-12-31 10:30:07,718 [NOTICE] - src/V4l2Device.cpp:147
        /dev/video0:YUYV size:640x480
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:229
        fps:1/25
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:230
        nbBuffer:2
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:117
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:119
        /dev/video10 support output
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:120
        /dev/video10 support capture
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:122
        /dev/video10 support read/write
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:123
        /dev/video10 support streaming
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2Device.cpp:147
        /dev/video10:H264 size:640x480
2018-12-31 10:30:07,760 [NOTICE] - src/V4l2MmapDevice.cpp:71
        Device /dev/video10 nb buffer:8
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:0 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:1 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:2 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:3 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:4 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:5 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:6 size:0
mmap: Invalid argument
2018-12-31 10:30:07,761 [INFO  ] - src/V4l2MmapDevice.cpp:90
        Device /dev/video10 buffer idx:7 size:0
mmap: Invalid argument
VIDIOC_STREAMON: Invalid argument
2018-12-31 10:30:07,762 [NOTICE] - src/v4l2compress_h264.cpp:136
        Start Capturing from /dev/video0
2018-12-31 10:30:07,762 [WARN  ] - src/v4l2compress_h264.cpp:163
        rc_method:1
2018-12-31 10:30:07,762 [WARN  ] - src/v4l2compress_h264.cpp:164
        i_qp_constant:23
2018-12-31 10:30:07,762 [WARN  ] - src/v4l2compress_h264.cpp:165
        f_rf_constant:23
x264 [info]: using cpu capabilities: ARMv6 NEON
x264 [info]: profile Constrained Baseline, level 3.0
2018-12-31 10:30:07,827 [NOTICE] - src/v4l2compress_h264.cpp:185
        Start Compressing /dev/video0 to /dev/video10
x264 [debug]: frame=   0 QP=20.00 NAL=3 Slice:I Poc:0   I:1200 P:0    SKIP:0    size=10642 bytes
2018-12-31 10:30:07,864 [WARN  ] - src/V4l2MmapDevice.cpp:226
        Device /dev/video10 buffer truncated available:0 needed:0
2018-12-31 10:30:07,864 [INFO  ] - src/v4l2compress_h264.cpp:239
        Copied nbnal:4 size:0
2018-12-31 10:30:07,865 [DEBUG ] - src/v4l2compress_h264.cpp:251
        dts:0 captureTime:0 convertTime:12 endodeTime:24 writeTime:0
x264 [debug]: frame=   1 QP=13.00 NAL=2 Slice:P Poc:2   I:0    P:40   SKIP:1160 size=2264 bytes
2018-12-31 10:30:07,896 [WARN  ] - src/V4l2MmapDevice.cpp:226
        Device /dev/video10 buffer truncated available:0 needed:0
2018-12-31 10:30:07,896 [INFO  ] - src/v4l2compress_h264.cpp:243
        Copied size:0
2018-12-31 10:30:07,896 [DEBUG ] - src/v4l2compress_h264.cpp:251
        dts:0 captureTime:0 convertTime:10 endodeTime:20 writeTime:0
x264 [debug]: frame=   2 QP=13.00 NAL=2 Slice:P Poc:4   I:0    P:15   SKIP:1185 size=109 bytes
2018-12-31 10:30:07,923 [WARN  ] - src/V4l2MmapDevice.cpp:226
        Device /dev/video10 buffer truncated available:0 needed:0
2018-12-31 10:30:07,923 [INFO  ] - src/v4l2compress_h264.cpp:243
        Copied size:0

mpromonet, 2019-01-05:
Hi,
The output buffer size is 0, something is going wrong.
Did you try to use the write interface instead of mmap, adding '-w'.
Best regards
Michel