mpromonet/v4l2tools #21: Build fails. Libyuv?

Siggemada, 2019-07-09:
Ok, so I tried to compile and get stuck on libyuv. What am I doing wrong?

orangepipc2:~/v4l2tools$ make
with log4cpp
git submodule init v4l2wrapper
Submodule 'v4l2wrapper' (https://github.com/mpromonet/v4l2wrapper) registered for path 'v4l2wrapper'
git submodule update v4l2wrapper
Cloning into '/home/hans/v4l2tools/v4l2wrapper'...
Submodule path 'v4l2wrapper': checked out '3fdaa04b94ec52a2e3f286103b72107cec2c1d32'
make -C v4l2wrapper
make[1]: Entering directory '/home/hans/v4l2tools/v4l2wrapper'
g++ -c -o src/V4l2MmapDevice.o src/V4l2MmapDevice.cpp -W -Wall -pthread -g -pipe  -I inc -DHAVE_LOG4CPP -I /usr/include
g++ -c -o src/V4l2Capture.o src/V4l2Capture.cpp -W -Wall -pthread -g -pipe  -I inc -DHAVE_LOG4CPP -I /usr/include
g++ -c -o src/V4l2Device.o src/V4l2Device.cpp -W -Wall -pthread -g -pipe  -I inc -DHAVE_LOG4CPP -I /usr/include
g++ -c -o src/V4l2Output.o src/V4l2Output.cpp -W -Wall -pthread -g -pipe  -I inc -DHAVE_LOG4CPP -I /usr/include
g++ -c -o src/logger.o src/logger.cpp -W -Wall -pthread -g -pipe  -I inc -DHAVE_LOG4CPP -I /usr/include
ar rcs libv4l2wrapper.a src/V4l2MmapDevice.o src/V4l2Capture.o src/V4l2Device.o src/V4l2Output.o src/logger.o
make[1]: Leaving directory '/home/hans/v4l2tools/v4l2wrapper'
mv v4l2wrapper/libv4l2wrapper.a .marcus.dahlberg@gmail.com
make -C v4l2wrapper clean
make[1]: Entering directory '/home/hans/v4l2tools/v4l2wrapper'
make[1]: Leaving directory '/home/hans/v4l2tools/v4l2wrapper'
g++ -o v4l2copy -W -Wall -pthread -g -pipe  -I include -DHAVE_LOG4CPP -I /usr/include -I v4l2wrapper/inc -DHAVE_JPEG src/v4l2copy.cpp libv4l2wrapper.a -llog4cpp  -ljpeg
git submodule init libyuv
Submodule 'libyuv' (https://github.com/openpeer/libyuv) registered for path 'libyuv'
git submodule update libyuv
Cloning into '/home/hans/v4l2tools/libyuv'...
Submodule path 'libyuv': checked out 'd204db647e591ccf0e2589236ecea90330d65a66'
cd libyuv && cmake . && make 
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found JPEG: /usr/lib/aarch64-linux-gnu/libjpeg.so  
CMake Warning (dev) at CMakeLists.txt:102 (if):
  Policy CMP0064 is not set: Support new TEST if() operator.  Run "cmake
  --help-policy CMP0064" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  TEST will be interpreted as an operator when the policy is set to NEW.
  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/hans/v4l2tools/libyuv
make[1]: Entering directory '/home/hans/v4l2tools/libyuv'
make[2]: Entering directory '/home/hans/v4l2tools/libyuv'
make[3]: Entering directory '/home/hans/v4l2tools/libyuv'
Scanning dependencies of target yuv
make[3]: Leaving directory '/home/hans/v4l2tools/libyuv'
make[3]: Entering directory '/home/hans/v4l2tools/libyuv'
[  2%] Building CXX object CMakeFiles/yuv.dir/source/compare.cc.o
[  5%] Building CXX object CMakeFiles/yuv.dir/source/compare_common.cc.o
[  7%] Building CXX object CMakeFiles/yuv.dir/source/compare_neon.cc.o
[ 10%] Building CXX object CMakeFiles/yuv.dir/source/compare_posix.cc.o
[ 13%] Building CXX object CMakeFiles/yuv.dir/source/compare_win.cc.o
[ 15%] Building CXX object CMakeFiles/yuv.dir/source/convert.cc.o
[ 18%] Building CXX object CMakeFiles/yuv.dir/source/convert_argb.cc.o
[ 21%] Building CXX object CMakeFiles/yuv.dir/source/convert_from.cc.o
[... 428 lines cut ...]
convert_argb.cc:(.text+0x154c): undefined reference to `NV12ToARGBRow_NEON'
libyuv.a(convert_argb.cc.o): In function `NV21ToARGB':
convert_argb.cc:(.text+0x16d4): undefined reference to `NV21ToARGBRow_NEON'
convert_argb.cc:(.text+0x16d8): undefined reference to `NV21ToARGBRow_NEON'
libyuv.a(convert_argb.cc.o): In function `M420ToARGB':
convert_argb.cc:(.text+0x184c): undefined reference to `NV12ToARGBRow_NEON'
convert_argb.cc:(.text+0x1850): undefined reference to `NV12ToARGBRow_NEON'
libyuv.a(convert_argb.cc.o): In function `YUY2ToARGB':
convert_argb.cc:(.text+0x1aa8): undefined reference to `YUY2ToARGBRow_NEON'
convert_argb.cc:(.text+0x1aac): undefined reference to `YUY2ToARGBRow_NEON'
libyuv.a(convert_argb.cc.o): In function `UYVYToARGB':
convert_argb.cc:(.text+0x1c48): undefined reference to `UYVYToARGBRow_NEON'
convert_argb.cc:(.text+0x1c4c): undefined reference to `UYVYToARGBRow_NEON'
collect2: error: ld returned 1 exit status
CMakeFiles/convert.dir/build.make:96: recipe for target 'convert' failed
make[3]: *** [convert] Error 1
make[3]: Leaving directory '/home/hans/v4l2tools/libyuv'
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/convert.dir/all' failed
make[2]: *** [CMakeFiles/convert.dir/all] Error 2
make[2]: Leaving directory '/home/hans/v4l2tools/libyuv'
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/hans/v4l2tools/libyuv'
Makefile:81: recipe for target 'libyuv.a' failed
make: *** [libyuv.a] Error 2

Siggemada, 2019-07-09:
Ok, so I swapped out "openpeer/obsolete.libyuv" for "webrtc-uwp/libyuv" and now libyuv builds. But build now fails at:

make v4l2compress_h264 VERBOSE=1
with log4cpp
g++ -o v4l2compress_h264 -W -Wall -pthread -g -pipe  -I include -DHAVE_LOG4CPP -I /usr/include -I v4l2wrapper/inc -DHAVE_JPEG src/v4l2compress_h264.cpp libyuv.a libv4l2wrapper.a -llog4cpp  -ljpeg -lx264 -I libyuv/include
src/v4l2compress_h264.cpp: In function ‘int main(int, char**)’:
src/v4l2compress_h264.cpp:203:28: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
       ConvertToI420((const uint8*)buffer, rsize,
                            ^~~~~
src/v4l2compress_h264.cpp:203:22: error: expected primary-expression before ‘const’
       ConvertToI420((const uint8*)buffer, rsize,
                      ^~~~~
src/v4l2compress_h264.cpp:203:22: error: expected ‘)’ before ‘const’
src/v4l2compress_h264.cpp:196:11: warning: unused variable ‘rsize’ [-Wunused-variable]
       int rsize = videoCapture->read(buffer, sizeof(buffer));
           ^~~~~
Makefile:112: recipe for target 'v4l2compress_h264' failed
make: *** [v4l2compress_h264] Error 1

Siggemada, 2019-07-09:
Added fpermissive but still:

 make v4l2compress_h264 VERBOSE=1
with log4cpp
g++ -o v4l2compress_h264 -fpermissive -W -Wall -pthread -g -pipe  -I include -DHAVE_LOG4CPP -I /usr/include -I v4l2wrapper/inc -DHAVE_JPEG src/v4l2compress_h264.cpp libyuv.a libv4l2wrapper.a -llog4cpp  -ljpeg -lx264 -I libyuv/include
src/v4l2compress_h264.cpp: In function ‘int main(int, char**)’:
src/v4l2compress_h264.cpp:203:28: warning: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
       ConvertToI420((const uint8*)buffer, rsize,
                            ^~~~~
src/v4l2compress_h264.cpp:203:22: error: expected primary-expression before ‘const’
       ConvertToI420((const uint8*)buffer, rsize,
                      ^~~~~
src/v4l2compress_h264.cpp:203:22: error: expected ‘)’ before ‘const’
src/v4l2compress_h264.cpp:196:11: warning: unused variable ‘rsize’ [-Wunused-variable]
       int rsize = videoCapture->read(buffer, sizeof(buffer));
           ^~~~~
Makefile:112: recipe for target 'v4l2compress_h264' failed
make: *** [v4l2compress_h264] Error 1

akavish12, 2020-01-23:
Any progress with the error? getting the same output.

Siggemada, 2020-01-24:
No, I never got it to build on ARM64. It does build on ARM32.

mpromonet, 2020-01-25:
Hi,
If you look to snapcraft build https://build.snapcraft.io/user/mpromonet/v4l2tools
arm64 is failing because libyuv doesnot build on arm64.
Maybe upgrading the libyuv source, or using different build option.
Best Regards,
Michel.

akavish12, 2020-01-26:
tried another source code, still getting the same error, tried using cmake, make, and gn tools, still no luck. any other idea?
thanks!

mpromonet, 2020-01-27:
Hi @akavish12,
What is the same error ? 
The error about NEON is from compiler, not link to make tools you are using.
Best Regards,
Michel.

akavish12, 2020-01-30:
[ 13%] Building CXX object CMakeFiles/yuv.dir/source/compare_win.cc.o
[ 15%] Building CXX object CMakeFiles/yuv.dir/source/convert.cc.o
[ 18%] Building CXX object CMakeFiles/yuv.dir/source/convert_argb.cc.o
[ 21%] Building CXX object CMakeFiles/yuv.dir/source/convert_from.cc.o
[ 23%] Building CXX object CMakeFiles/yuv.dir/source/convert_from_argb.cc.o
[ 26%] Building CXX object CMakeFiles/yuv.dir/source/convert_jpeg.cc.o
g++ -c -o src/logger.o src/logger.cpp -W -Wall -pthread -g -pipe  -I inc -DHAVE_LOG4CPP -I /usr/include
[ 28%] Building CXX object CMakeFiles/yuv.dir/source/convert_to_argb.cc.o
[ 31%] Building CXX object CMakeFiles/yuv.dir/source/convert_to_i420.cc.o
[ 34%] Building CXX object CMakeFiles/yuv.dir/source/cpu_id.cc.o
[ 36%] Building CXX object CMakeFiles/yuv.dir/source/format_conversion.cc.o
[ 39%] Building CXX object CMakeFiles/yuv.dir/source/mjpeg_decoder.cc.o
[ 42%] Building CXX object CMakeFiles/yuv.dir/source/mjpeg_validate.cc.o
[ 44%] Building CXX object CMakeFiles/yuv.dir/source/planar_functions.cc.o
ar rcs libv4l2wrapper.a src/V4l2Access.o src/V4l2MmapDevice.o src/V4l2Capture.o src/V4l2Device.o src/V4l2ReadWriteDevice.o src/V4l2Output.o src/logger.o
make[1]: Leaving directory '/root/v4l2tools/v4l2wrapper'
mv v4l2wrapper/libv4l2wrapper.a .
make -C v4l2wrapper clean
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/root/v4l2tools/v4l2wrapper'
make[1]: Leaving directory '/root/v4l2tools/v4l2wrapper'
g++ -o v4l2copy -W -Wall -pthread -g -pipe  -I include -DHAVE_LOG4CPP -I /usr/include -I v4l2wrapper/inc -I /opt/vc/include/ -I /opt/vc/include/interface/vcos/ -I /opt/vc/include/interface/vcos/pthreads/ -I /opt/vc/include/interface/vmcs_host/linux/ -I /opt/vc/src/hello_pi/libs/ilclient  -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -DHAVE_JPEG src/v4l2copy.cpp libv4l2wrapper.a -llog4cpp  -L /opt/vc/lib -L /opt/vc/src/hello_pi/libs/ilclient -lpthread -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -ljpeg
[ 47%] Building CXX object CMakeFiles/yuv.dir/source/rotate.cc.o
[ 50%] Building CXX object CMakeFiles/yuv.dir/source/rotate_argb.cc.o
[ 52%] Building CXX object CMakeFiles/yuv.dir/source/rotate_mips.cc.o
[ 55%] Building CXX object CMakeFiles/yuv.dir/source/rotate_neon.cc.o
[ 57%] Building CXX object CMakeFiles/yuv.dir/source/row_any.cc.o
[ 60%] Building CXX object CMakeFiles/yuv.dir/source/row_common.cc.o
g++ -o v4l2source_yuv -W -Wall -pthread -g -pipe  -I include -DHAVE_LOG4CPP -I /usr/include -I v4l2wrapper/inc -I /opt/vc/include/ -I /opt/vc/include/interface/vcos/ -I /opt/vc/include/interface/vcos/pthreads/ -I /opt/vc/include/interface/vmcs_host/linux/ -I /opt/vc/src/hello_pi/libs/ilclient  -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -DHAVE_JPEG src/v4l2source_yuv.cpp libv4l2wrapper.a -llog4cpp  -L /opt/vc/lib -L /opt/vc/src/hello_pi/libs/ilclient -lpthread -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -ljpeg
[ 63%] Building CXX object CMakeFiles/yuv.dir/source/row_mips.cc.o
[ 65%] Building CXX object CMakeFiles/yuv.dir/source/row_neon.cc.o
[ 68%] Building CXX object CMakeFiles/yuv.dir/source/row_posix.cc.o
[ 71%] Building CXX object CMakeFiles/yuv.dir/source/row_win.cc.o
[ 73%] Building CXX object CMakeFiles/yuv.dir/source/scale.cc.o
[ 76%] Building CXX object CMakeFiles/yuv.dir/source/scale_argb.cc.o
[ 78%] Building CXX object CMakeFiles/yuv.dir/source/scale_common.cc.o
[ 81%] Building CXX object CMakeFiles/yuv.dir/source/scale_mips.cc.o
[ 84%] Building CXX object CMakeFiles/yuv.dir/source/scale_neon.cc.o
g++ -o v4l2grab_h264 src/encode_omx.cpp src/v4l2grab_h264.cpp /opt/vc/src/hello_pi/libs/ilclient/libilclient.a libv4l2wrapper.a -W -Wall -pthread -g -pipe  -I include -DHAVE_LOG4CPP -I /usr/include -I v4l2wrapper/inc -I /opt/vc/include/ -I /opt/vc/include/interface/vcos/ -I /opt/vc/include/interface/vcos/pthreads/ -I /opt/vc/include/interface/vmcs_host/linux/ -I /opt/vc/src/hello_pi/libs/ilclient  -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -DHAVE_JPEG -llog4cpp  -L /opt/vc/lib -L /opt/vc/src/hello_pi/libs/ilclient -lpthread -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -ljpeg 
[ 86%] Building CXX object CMakeFiles/yuv.dir/source/scale_posix.cc.o
[ 89%] Building CXX object CMakeFiles/yuv.dir/source/scale_win.cc.o
[ 92%] Building CXX object CMakeFiles/yuv.dir/source/video_common.cc.o
[ 94%] Linking CXX static library libyuv.a
make[3]: Leaving directory '/root/v4l2tools/libyuv'
[ 94%] Built target yuv
make[3]: Entering directory '/root/v4l2tools/libyuv'
Scanning dependencies of target convert
make[3]: Leaving directory '/root/v4l2tools/libyuv'
make[3]: Entering directory '/root/v4l2tools/libyuv'
[ 97%] Building CXX object CMakeFiles/convert.dir/util/convert.cc.o
[100%] Linking CXX executable convert
src/encode_omx.cpp: In function ‘ILCLIENT_T* encode_init(COMPONENT_T**)’:
[... cut ...]
convert_argb.cc:(.text+0x1a58): undefined reference to `YUY2ToARGBRow_NEON'
/usr/bin/ld: convert_argb.cc:(.text+0x1a5c): undefined reference to `YUY2ToARGBRow_NEON'
/usr/bin/ld: libyuv.a(convert_argb.cc.o): in function `UYVYToARGB':
convert_argb.cc:(.text+0x1bf4): undefined reference to `UYVYToARGBRow_NEON'
/usr/bin/ld: convert_argb.cc:(.text+0x1bf8): undefined reference to `UYVYToARGBRow_NEON'

same error as far as i understand like Siggemada had, tried many methods, can you be more specific on how to resolve this issue (if it's even possible?)? 

Thanks alot!

mpromonet, 2020-01-30:
Hi akavish12,
Did you tried to build https://chromium.googlesource.com/libyuv/libyuv for your platform ?
What I understand is that it doesnot build, then my advice is to resolve it in this project.
Then it will be possible to upgrade the submodule link.
Best Regards,
MicheL

akavish12, 2020-01-30:
tried to build it using the same link, still getting the same error.. :( thanks for your time.

akavish12, 2020-01-30:
Don't know if that helps, but i'm trying to build it on raspberry pi 3B+ (64 bit) running Kali - For the driver support.
on x86_64 PC it builds great in seconds.
Anyway thanks again for your time.

akavish12, 2020-02-01:
I found a version that does compile using autogen: https://github.com/lemenkov/libyuv/tree/autotools < that specific version.
how do i modify the Makefile correctly to proceed?
thanks again..

akavish12, 2020-02-01:
i've commented out the libyuv section in makefile, copied manually "libyuv.a" to "v4l2tools" directory, and run make again, now it compiles, getting another error "/usr/bin/ld: ./.libs/libh264bitstream.so: undefined reference to `debug_pps'", trying to use another source.

akavish12, 2020-02-15:
So, after many tries i've managed to compile and make it work.
here's how:
clone this repo: git clone https://github.com/mpromonet/v4l2tools
remove libyuv directory.
manually clone this repo: git clone -b autotools https://github.com/lemenkov/libyuv
cd into libyuv dir.
run ./autogen.sh
run ./configure --with-mjpeg
run make
run make install (sudo)
copy .libs/libyuv.a to v4ltools directory.
edit Makefile:
libyuv.a: (comment out this section as follows):
#        git submodule init libyuv
#       git submodule update libyuv
#        cd libyuv && cmake . && make 
#       mv libyuv/libyuv.a .
#        make -C libyuv clean

run make
run make install

now it works on arm64 (aarch64) as for RaspberryPi (64 bit) Rock64, Odroid C2 and others.

Enjoy :)

mpromonet, 2020-05-20:
Hi akavish12,

I think this is now fixed building libyuv with the flag `-DLIBYUV_DISABLE_NEON` (see commit https://github.com/mpromonet/v4l2tools/commit/41b152f4dcc25c6e9d67b9d090681cc7861d55b7)

Best Regards,
Michel.

closed 2020-05-24 by mpromonet