mpromonet/v4l2tools #25: H.264 Baseline wanted

Serveurperso, 2019-09-20:
Hello

I need the Baseline profile for H.264
How to change the H.264 configuration ?

Thanks
Pascal

Serveurperso, 2019-09-20:
I try to do this work.

Inside v4l2grab_h264.cpp
There is hardcoded parameters like bitrate set to 10000000 (i need less!)

encode_config_output(video_encode, OMX_VIDEO_CodingAVC, 10000000);

I also need to modofy encode_config_input() to add Profile setting.

I foud some documentation about OMX API...

http://maemo.org/api_refs/5.0/beta/libomxil-bellagio/_o_m_x___video_8h-source.html

OMX_VIDEO_H263PROFILETYPE
OMX_VIDEO_H263ProfileBaseline            = 0x01,

Serveurperso, 2019-09-20:
```
+       // set baseline profile
+       OMX_VIDEO_PARAM_PROFILELEVELTYPE profileLevel;
+       memset(&profileLevel, 0, sizeof(OMX_VIDEO_PARAM_PROFILELEVELTYPE));
+       profileLevel.nSize = sizeof(OMX_VIDEO_PARAM_PROFILELEVELTYPE);
+       profileLevel.nVersion.nVersion = OMX_VERSION;
+       profileLevel.nPortIndex = 201;
+       profileLevel.eProfile = OMX_VIDEO_AVCProfileBaseline;
+       profileLevel.eLevel = OMX_VIDEO_AVCLevel4;
+
+       fprintf(stderr, "OMX_SetParameter(OMX_IndexParamVideoProfileLevelCurrent) for video_encode:201...\n");
+       omx_return = OMX_SetParameter(ILC_GET_HANDLE(handle), OMX_IndexParamVideoProfileLevelCurrent, &profileLevel);
+       if (omx_return != OMX_ErrorNone)
+       {
+               fprintf(stderr, "%s:%d: OMX_SetParameter() for video_encode port 201 failed with %x!\n", __FUNCTION__, __LINE__, omx_return);
+               return false;
+       }
+
```

mpromonet, 2019-09-21:
Hi Pascal,

It seems nice to add a parameter to choose the H264 profile.
Feel free to submit a pull request.

Best Regards,
Michel.

Serveurperso, 2019-09-21:
Yes this is the idea, it's nice to add some H.264 configurable parameters.

But before I have a blocking problem I can't launch any v4l2tools without the "-w" parameter else I get "Inappropriate ioctl for device"

And with this "-w" I get a lot of strange buffering latency I don't understand where is the problem:(

Here (no ads) https://www.vigibot.com you can select and move few raspberry PI robot with /dev/video0 from camera as source and there is almost no latency, like 0.1sec from end to end (robot to browser) I would to add a /dev/video1 for remote console display ! Or for duplicate a local raw camera stream which will be used in conjunction with the low latency H.264 hardware encoded stream.

mpromonet, 2019-09-21:
Hi Pascal,
Usually this is because a library like libuv4lext.so or other are preloaded.
However using v4l2loopback I have some problems to restart.
Best Regards,
Michel.

Serveurperso, 2019-09-30:
```
root@raspberrypi:~# lsmod
Module                  Size  Used by
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
evdev                  24576  1
brcmfmac              311296  0
snd_usb_audio         192512  1
brcmutil               16384  1 brcmfmac
snd_hwdep              16384  1 snd_usb_audio
snd_usbmidi_lib        32768  1 snd_usb_audio
snd_rawmidi            32768  1 snd_usbmidi_lib
snd_seq_device         16384  1 snd_rawmidi
sha256_generic         20480  0
raspberrypi_hwmon      16384  0
cfg80211              614400  1 brcmfmac
hwmon                  16384  1 raspberrypi_hwmon
snd_bcm2835            24576  1
rfkill                 28672  4 cfg80211
snd_pcm               102400  2 snd_usb_audio,snd_bcm2835
snd_timer              32768  1 snd_pcm
bcm2835_codec          36864  0
bcm2835_v4l2           45056  0
snd                    73728  12 snd_hwdep,snd_usb_audio,snd_timer,snd_rawmidi,snd_usbmidi_lib,snd_seq_device,snd_bcm2835,snd_pcm
v4l2_mem2mem           24576  1 bcm2835_codec
bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
v4l2_common            16384  1 bcm2835_v4l2
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_dma_contig    20480  1 bcm2835_codec
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         24576  3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
i2c_bcm2835            16384  0
videobuf2_common       45056  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              200704  6 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
media                  36864  2 videodev,v4l2_mem2mem
vc_sm_cma              36864  1 bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
i2c_dev                20480  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  450560  31
```

commit 5fa1d8d0 (2020-05-20): fix #25: allow to set profile/level of H264 compression
closed 2020-05-20 by mpromonet