mpromonet/v4l2tools #27: Can't make v4l2grab_h264 to work properly

brainstormi, 2019-10-27:
Hi...
   I'm trying to get v4l2grab_h264 to capture desktop and stream it with v4l2rtspstreamserver (running in Pi3B/osmc/kodi) without luck, maybe I'm not using it correctly. I created a loopback device with:

_sudo modprobe v4l2loopback video_nr=10_
v4l2grab_h264 --vv /dev/video10 
and stream it with:
v4l2rtspsetver --vv /dev/video10

I'm trying to reproduce it with VLC (rtsp://ip:8554/unicast) without luck. Most of the times remains there forver without displaying anything and from time to time it displays a static image of the desktop with the bottom of the screem totally distorted.

I tried to do the same using raspidisp driver from uv4l project and v4l2compress_omx with a lot more luck, making easy to connect 90% of the times and when it does it plays fluently the desktop with some artifacts, but I'll open a different issue about it.

I played with v4l2grab_h264 write API (-w) and v4l2rtspserver (-r and -w) in all combinations possible without success.

Below the logs from v4l2grab_h264 and v4l2rtsp from the initial example, frame size are already suspicious:

**v4l2grab_h264 output**
[(https://pastebin.com/5LSD4Mi8]
)
**v4l2rtspserver -vv /dev/video10**
[https://pastebin.com/CUP10iVE]

Regards,

brainstormi, 2019-10-27:
Playing a movie in full screen v4l2grab_h264 shows in the log frame sizes more logical, and in VLC you seen screen is updated more frequently although totally garbled, full of artifacts:

[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:65536
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:65536
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:11412
[WARN] src/v4l2grab_h264.cpp:184
	fwrite: Error emptying buffer:4294967295!=54068
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:1605
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:44765
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:46676
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:48937
[DEBUG] src/v4l2grab_h264.cpp:188
	Writing frame size:13390

mpromonet, 2019-10-27:
Hi,
This commit https://github.com/mpromonet/v4l2tools/commit/79af89894392d3cdef14cf3f71a8eb2662fdb006#diff-7a50a337891585a439948f5e4f0959f5 should help.
Best Regards,
Michel.

brainstormi, 2019-10-27:
A lot better now!!. Now it's possible to connect to RTSP stream without hassle and it shows a live desktop with the similar artifacts issues reported when using raspidisp driver  #28  . 
The issue now is that it's not possible to stream a movie in fullscreen. For example when trying to do it with kodi, v4l2grab_h264 just freeze when starting the movie or just a few seconds later (5-10 seconds). This doesn't happen with uv4l driver.
There is no error in the output console, it just stops. Only way to resume it is killing the process and launching it again:

	Writing frame size:23275
[DEBUG] src/v4l2grab_h264.cpp:183
	Writing frame size:5266
[DEBUG] src/v4l2grab_h264.cpp:183
	Writing frame size:55524
[DEBUG] src/v4l2grab_h264.cpp:183
	Writing frame size:38644
[DEBUG] src/v4l2grab_h264.cpp:183
	Writing frame size:21844
[DEBUG] src/v4l2grab_h264.cpp:183

brainstormi, 2019-10-27:
Great!. Your last commit seems to have fixed the streaming issues with the grabber, including displaying movies in full screen. 
There is a minor issue, also present in raspidisp. In both cases if you configure Kodi, to sync screen refresh rate with movie when it starts/stops, kills the grabber with the following error:

```
[DEBUG] src/v4l2grab_h264.cpp:183
v4l2grab_h264: src/v4l2grab_h264.cpp:46: int take_snapshot(void*, DISPMANX_DISPLAY_HANDLE_T, DISPMANX_RESOURCE_HANDLE_T, DISPMANX_MODEINFO_T, VC_RECT_T*, DISPMANX_TRANSFORM_T): Assertion `ret == 0' failed.
	Writing frame size:753Abortado

```

The only program I tested using the same dispmanx API that manages properly this scenario is dispmanx_vncserver, in particular this fork: [(https://github.com/patrikolausson/dispmanx_vnc)], but taking a look at it, I could understand that a change of resolution could cause an issue with `vc_dispmanx_snapshot(display,resource,transform)` because the resource created has change its size,  but not with refresh rate.