mpromonet/v4l2tools #28: v4l2compress_omx artifacts at the bottom of the screen (raspidisp)

brainstormi, 2019-10-27:
Hi...
    First of all this issue is reported using raspidisp (uv4l) driver because I can't make v4l2grab_h264 to work properly, so If you consider this is not supported please don't hesitate to close the issue.
    When running v4l2compress_omx over v4l2loopback device from /dev/video0  output generated by raspidisp, desktop is streamed smoothly although with a delay about 4 seconds for a 720p desktop. Issue is that frequently shows completely garbled output during few seconds ath ike 20%-30% of the bottom screen, like showing a distorted static image from a previous scene. This happens more frequently when the scene you are seeing changes, for example in a playing a movie. I tried to modify the source code, for example increasing the bitrate with the same outcome. Also tried to play with h264 profiles in the code without luck to get a stream.
    In the case of using v4l2compress_h264 this doesn't happens,  but as I'm running  a pi3b I'm already maxing 1 of the cores, dropping frames... not much, it's can be viewed quite well. Issue here is that trying to open the stream from VLC it's not reliable, sometimes take too much time or doesn't open and you're forced to try multiple times until it works.
   I tried to increase the number of threads in v4l2compress_h264.cpp from 1 to 2 or 4 to avoid CPU spike, but then video quality gets totally affected, showing a similar pattern like the one with v4l2compress_omx but a lot worst, affecting 50% or more of the screen at the top and/or bottom.
Regards,

brainstormi, 2019-10-27:
Regarding v4l2compress_omx it's frequently repeated this error in console (using raspidisp driver):

[DEBUG] src/v4l2compress_omx.cpp:237
	Not getting it Error sending buffer for filling: 80001005

mpromonet, 2019-10-27:
Hi brainstormi,
There is an error when out is NULL, it should not try to fillit.
I made some test with v4l2grab it seems there is also some problem, I did not try these tools since a while.
Best Regards,
Michel.

brainstormi, 2019-10-27:
Not sure if the video issues at the bottom of the screen could be related to this issue reported in the past with gstreamer omx encoding:
[(https://www.raspberrypi.org/forums/viewtopic.php?t=200306)]
I did a quick try trying to increase port buffer size to 512k as 6by9 suggested but didn't fixed the problem. It seems issue could be related to the fact that a frame could be splitted over multiple buffers, but only the last is flagged with OMX_BUFFERFLAG_ENDOFFRAME, being needed to look for it before rebuilding the full frame.

nschimme, 2020-05-10:
I just wanted to add that using a 640x480 resolution makes the corruption disappear most of the time. It becomes very evident at higher resolutions.

Tuckie, 2020-05-28:
@nschimme I'm having the same issue, how did you set the resolution?

nschimme, 2020-05-30:
You need to `sudo apt install v4l-utils` and then use the [v4l2-ctl](https://manpages.ubuntu.com/manpages/bionic/man1/v4l2-ctl.1.html) program.

I've placed this in my `/etc/rc.local`:

`v4l2-ctl -d /dev/video0 -v width=640,height=480`

Tuckie, 2020-06-03:
Thank you! I'll give that a shot.

nschimme, 2020-06-17:
FYI, this might be related but I am on kernel 4.19.118 and it didn't resolve it.

https://github.com/raspberrypi/firmware/issues/1108

closed 2021-10-03 by mpromonet