mpromonet/v4l2tools #32: Using -W and -H with v4l2copy

mgoeller, 2019-12-12:
Hi, how can I tell v4l2copy which resolution and which codec to copy from /dev/video0? If I want to copy the raspberry pi camera input with ffmpeg I can specify 1280x720 as resolution, using v4l2copy results in 1024x768. The -W -H parameters don't seem to be recognised.

```
    pi@raspberrypi:~ $ v4l2copy -vv -W 1280 -H 720 /dev/video0 /dev/video7
    v4l2copy: invalid option -- 'W'
    v4l2copy: invalid option -- 'H'
    log level:700
    
    [NOTICE] src/V4l2MmapDevice.cpp:49
    VIDIOC_REQBUFS: Inappropriate ioctl for device
        Device 1280
    [NOTICE] src/V4l2MmapDevice.cpp:141
    VIDIOC_STREAMOFF: Inappropriate ioctl for device
    VIDIOC_REQBUFS: Inappropriate ioctl for device
    	Device 1280
    [WARN] src/v4l2copy.cpp:91
    	Cannot create V4L2 capture interface for device:1280
```

mpromonet, 2019-12-12:
Hi mgoller,

In fact v4l2copy just copy the frame, it doesnot scale or crop the input frame. There is no -W/-H options.

Best Regards,
Michel.

mgoeller, 2019-12-16:
But one can grab a Raspberry Pi camera frame from /dev/video0 with way more than 1024x768 (up to the 5MP of the camera). So somehow I should be able to specify what to grab, no? Or how does it end up grabbing automatically the 1024x768 resolution?

closed 2020-02-15 by mpromonet