mpromonet/v4l2tools #33: v4l2compress_omx, Mandatory capability not available for device:/dev/video1

imchesterhk, 2020-05-22:
Hi Michel,

Is me again sorry, i think i should raise this as a question instead of a issue but i dont know how, i guess it is just my problem.
After i successfully create the v4l2rtspserver with my pi, i would like to try convert my YUY2 capture device to H264 and stream it out with v4l2rtspserver, but i got following error
Mandatory capability not available for device:/dev/video1
I am not sure about this, is it my device problem ? or i miss some steps ?
same, i am using Pi 4B, merci beaucoup

```
pi@raspberrypi:~ $ v4l2compress_omx /dev/video0 /dev/video1
2020-05-22 21:11:31,940 [NOTICE] - src/V4l2Device.cpp:123
        driver:uvcvideo capabilities:84a00001 mandatory:4000001
2020-05-22 21:11:31,941 [NOTICE] - src/V4l2Device.cpp:126
        /dev/video0 support capture
2020-05-22 21:11:31,941 [NOTICE] - src/V4l2Device.cpp:129
        /dev/video0 support streaming
2020-05-22 21:11:31,941 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video0:MJPG size:1920x1080 bufferSize:4147200
2020-05-22 21:11:33,261 [NOTICE] - src/V4l2Device.cpp:205
        /dev/video0:MJPG size:1920x1080 bufferSize:4147200
2020-05-22 21:11:33,261 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video0:MJPG size:1920x1080 bufferSize:4147200
2020-05-22 21:11:33,261 [NOTICE] - src/V4l2MmapDevice.cpp:49
        Device /dev/video0
2020-05-22 21:11:33,314 [NOTICE] - src/V4l2MmapDevice.cpp:73
        Device /dev/video0 nb buffer:10
2020-05-22 21:11:33,363 [NOTICE] - src/V4l2Device.cpp:123
        driver:uvcvideo capabilities:84a00001 mandatory:4000002
2020-05-22 21:11:33,363 [NOTICE] - src/V4l2Device.cpp:126
        /dev/video1 support capture
2020-05-22 21:11:33,363 [NOTICE] - src/V4l2Device.cpp:129
        /dev/video1 support streaming
2020-05-22 21:11:33,364 [ERROR ] - src/V4l2Device.cpp:135
        Mandatory capability not available for device:/dev/video1
2020-05-22 21:11:33,364 [ERROR ] - src/V4l2Device.cpp:73
        Cannot init device:/dev/video1
2020-05-22 21:11:33,364 [NOTICE] - src/V4l2MmapDevice.cpp:141
        Device /dev/video1
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
2020-05-22 21:11:33,364 [WARN  ] - src/v4l2compress_omx.cpp:127
        Cannot create V4L2 output interface for device:/dev/video1
2020-05-22 21:11:33,364 [NOTICE] - src/V4l2MmapDevice.cpp:141
        Device /dev/video0
pi@raspberrypi:~ $

```

```
pi@raspberrypi:~ $ v4l2compress
2020-05-22 21:21:56,938 [NOTICE] - src/V4l2Device.cpp:123
        driver:uvcvideo capabilities:84a00001 mandatory:4000001
2020-05-22 21:21:56,938 [NOTICE] - src/V4l2Device.cpp:126
        /dev/video0 support capture
2020-05-22 21:21:56,939 [NOTICE] - src/V4l2Device.cpp:129
        /dev/video0 support streaming
2020-05-22 21:21:56,939 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video0:YUYV size:720x480 bufferSize:691200
2020-05-22 21:21:58,258 [NOTICE] - src/V4l2Device.cpp:205
        /dev/video0:YUYV size:720x480 bufferSize:691200
2020-05-22 21:21:58,258 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video0:YUYV size:720x480 bufferSize:691200
2020-05-22 21:21:58,258 [NOTICE] - src/V4l2MmapDevice.cpp:49
        Device /dev/video0
2020-05-22 21:21:58,269 [NOTICE] - src/V4l2MmapDevice.cpp:73
        Device /dev/video0 nb buffer:10
2020-05-22 21:21:58,292 [NOTICE] - src/V4l2Device.cpp:123
        driver:uvcvideo capabilities:84a00001 mandatory:4000002
2020-05-22 21:21:58,292 [NOTICE] - src/V4l2Device.cpp:126
        /dev/video1 support capture
2020-05-22 21:21:58,292 [NOTICE] - src/V4l2Device.cpp:129
        /dev/video1 support streaming
2020-05-22 21:21:58,292 [ERROR ] - src/V4l2Device.cpp:135
        Mandatory capability not available for device:/dev/video1
2020-05-22 21:21:58,293 [ERROR ] - src/V4l2Device.cpp:73
        Cannot init device:/dev/video1
2020-05-22 21:21:58,293 [NOTICE] - src/V4l2MmapDevice.cpp:141
        Device /dev/video1
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
2020-05-22 21:21:58,293 [WARN  ] - src/v4l2compress.cpp:54
        Cannot create V4L2 output interface for device:/dev/video1
2020-05-22 21:21:58,293 [NOTICE] - src/V4l2MmapDevice.cpp:141
        Device /dev/video0
pi@raspberrypi:~ $

```

mpromonet, 2020-05-22:
Hi imchesterhk,

It's seems you have 2 camera /dev/video0 and /dev/video1 ? Both are capture device.

These smalls compression tools, read from V4L2 capture device and write to a V4L2 output device.
The output device maybe a loopback one (for instance created by v4l2loopback).

Best Regards,
Michel.

imchesterhk, 2020-05-22:
So is it i can use v4l2loopback to create a virtual video output device and have the v4l2compress_omx write the video to the new virtual video device ? If so i will try to figure it out how to do it. you may close this issue. Thanks again

imchesterhk, 2020-05-31:
After i installed v4l2loopback i am able to use v4l2compress_omx now.

```
modprobe v4l2loopback video_nr=20
v4l2compress_omx /dev/video0 /dev/video20
```

Set the virtual video device to 20 and read the actual device from 0 and output to 20

Thank you !

closed 2020-05-31 by imchesterhk