mpromonet/v4l2tools #34: Is there a option can manually set the input resolutions size with v4l2compress_omx ?

imchesterhk, 2020-05-31:
Hi,

When i am using the command v4l2compress_omx   ,seem it will just use the default input size. My capture card maximum input size is 1920X1080, is there a option i can add to use ? below is the command when i run v4l2compress_omx, it just read only 720x480. thank you, 

```
pi@raspberrypi:~ $ v4l2compress_omx /dev/video0 /dev/video20
2020-06-01 03:58:58,210 [NOTICE] - src/V4l2Device.cpp:123
        driver:uvcvideo capabilities:84a00001 mandatory:4000001
2020-06-01 03:58:58,210 [NOTICE] - src/V4l2Device.cpp:126
        /dev/video0 support capture
2020-06-01 03:58:58,211 [NOTICE] - src/V4l2Device.cpp:129
        /dev/video0 support streaming
2020-06-01 03:58:58,211 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video0:YUYV size:720x480 bufferSize:691200
2020-06-01 03:58:59,529 [NOTICE] - src/V4l2Device.cpp:205
        /dev/video0:YUYV size:720x480 bufferSize:691200
2020-06-01 03:58:59,529 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video0:YUYV size:720x480 bufferSize:691200
2020-06-01 03:58:59,530 [NOTICE] - src/V4l2MmapDevice.cpp:49
        Device /dev/video0
2020-06-01 03:58:59,541 [NOTICE] - src/V4l2MmapDevice.cpp:73
        Device /dev/video0 nb buffer:10
2020-06-01 03:58:59,560 [NOTICE] - src/V4l2Device.cpp:123
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
2020-06-01 03:58:59,560 [NOTICE] - src/V4l2Device.cpp:125
        /dev/video20 support output
2020-06-01 03:58:59,560 [NOTICE] - src/V4l2Device.cpp:126
        /dev/video20 support capture
2020-06-01 03:58:59,560 [NOTICE] - src/V4l2Device.cpp:128
        /dev/video20 support read/write
2020-06-01 03:58:59,561 [NOTICE] - src/V4l2Device.cpp:129
        /dev/video20 support streaming
2020-06-01 03:58:59,561 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video20:H264 size:720x480 bufferSize:1382400
2020-06-01 03:58:59,561 [NOTICE] - src/V4l2Device.cpp:205
        /dev/video20:H264 size:720x480 bufferSize:1382400
2020-06-01 03:58:59,561 [NOTICE] - src/V4l2Device.cpp:61
        /dev/video20:H264 size:720x480 bufferSize:1382400
2020-06-01 03:58:59,561 [NOTICE] - src/V4l2MmapDevice.cpp:49
        Device /dev/video20
2020-06-01 03:58:59,561 [NOTICE] - src/V4l2MmapDevice.cpp:73
        Device /dev/video20 nb buffer:2
2020-06-01 03:58:59,563 [NOTICE] - src/v4l2compress_omx.cpp:131
        Start Capturing from /dev/video0
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 160x64 160x64 @1966080 20
Port 200: in 1/1 15360 16 disabled,not pop.,not cont. 720x480 720x480 @1966080 20
OMX_SetParameter(OMX_IndexParamVideoPortFormat) for video_encode:201...
OMX_SetParameter(OMX_IndexParamBrcmVideoAVCInlineHeaderEnable) for video_encode:201...
OMX_SetParameter(OMX_IndexParamVideoProfileLevelCurrent) for video_encode:201...
encode to idle...
enabling port buffers for 200...
enabling port buffers for 201...
encode to executing...
2020-06-01 03:58:59,575 [NOTICE] - src/v4l2compress_omx.cpp:153
        Start Compressing /dev/video0 to /dev/video20

```

imchesterhk, 2020-05-31:
Also, may i have more information about -p -l ? is it where i should do the setting ?

is it the same level about this ?
[https://en.wikipedia.org/wiki/Advanced_Video_Coding#Levels](url)

```
pi@raspberrypi:~ $ v4l2compress_omx -h
v4l2compress_omx [-v[v]] source_device dest_device
         -v            : verbose
         -vv           : very verbose
         -r            : V4L2 capture using read interface (default use memory mapped buffers)
         -w            : V4L2 capture using write interface (default use memory mapped buffers)
         -p profile    : H264 profile (default 8)
         -l level      : H264 level (default 2048)
         source_device : V4L2 capture device (default /dev/video0)
         dest_device   : V4L2 capture device (default /dev/video1)

```

mpromonet, 2020-06-01:
Hi imchesterhk ,

v4l2compress_omx just compress, it takes the size of the input and use it for the output.
Then to change the resolution you should set it before running it, it will use it.

And yes profiles and level are describe in the link https://en.wikipedia.org/wiki/Advanced_Video_Coding#Levels & https://en.wikipedia.org/wiki/Advanced_Video_Coding#Profiles. You can get the acceptable values from https://github.com/mpromonet/v4l2tools/blob/master/include/encode_omx.h

Best Regards,
Michel.

closed 2020-07-04 by mpromonet