mpromonet/v4l2tools #35: v4l2compress encodes as fast as possible instead of at device FPS

gabenell, 2020-09-23:
I noticed that v4l2compress uses 100% CPU when running, and that when I serve the output of v4l2compress using v4l2rtspserver, it sends hundreds of FPS. My video source is a v4l2loopback device.

The problem seems to be the while loop in v4l2compress.cpp, which encodes as fast as possible. I verified by adding a sleep that the CPU usage dropped and the FPS received by the RTSP client went down to a sane rate.

Is there any way to make v4l2compress run at the native framerate of the source device? Trying to achieve this by sleeping is not ideal.

flixman, 2021-07-24:
@gabenell I find myself on this same situation. Did you manage to get it fixed? Where did you put the sleep, as a fallback option?

fplanque, 2024-03-24:
Even when using a USB camera as source, how would one decide to compress a 15fps stream instead of the max 30fps stream?

There is no framerate option for v4l2compress nor v4l2compress_omx.

Also I don't see how I could modify the device framerate with v4l2-ctl ... ?

Thank you.

mpromonet, 2024-03-24:
Hi @fplanque 

This tools are not setting fps, you should set it using v4l2-ctl --set-parm.

Best Regards,
Michel.