mpromonet/v4l2tools #38: Use v4l2copy to clone /dev/video0 to multipe /dev/video<x>

FraBle, 2021-04-28:
Is it possible to use `v4l2copy` to mirror/clone the video feed from `/dev/video0` to multiple `v4l2loopback` devices?

Example using ffmpeg:

```bash
ffmpeg -framerate 30 -f v4l2 -i /dev/video0 -codec copy -f v4l2 /dev/video1 -codec copy -f v4l2 /dev/video2
```

mpromonet, 2021-04-28:
Hi @FraBle,

I guess you can do this running `v4l2copy /dev/video0 /dev/video1` and  `v4l2copy /dev/video1 /dev/video2`.
However v4l2loopback support multiple consumer, so maybe using only one v4l2loopbakc is enouph.
It is also easy to modify v4l2copy to copy to a list a output.

Best Regards,
Michel.

FraBle, 2021-04-29:
> However v4l2loopback support multiple consumers, so maybe using only one v4l2loopbakc is enough.

TIL 💡

closed 2021-04-29 by FraBle