mpromonet/v4l2tools #39: H264 failing with v4l2loopback

owen7900, 2021-08-03:
When running v4l2 compress in ubuntu 20.04 with v4l2loopback installed from apt I get an error saying `Cannot set pixelformat to:h264 format is:BGR4`. I have put the full output of the command below. 

Thanks in advance
`
./v4l2compress -vv -fh264 /dev/video10 /dev/video11                                                                                                           
log level:700                                                                                                                                                                                                      
                                                                                                                                                                                                                   
[NOTICE] src/V4l2Device.cpp:133                                                                                                                                                                                    
        driver:v4l2 loopback capabilities:85208003 mandatory:4000001                                                                                                                                               
[NOTICE] src/V4l2Device.cpp:135                                                                                                                                                                                    
        /dev/video10 support output                                                                                                                                                                                
[NOTICE] src/V4l2Device.cpp:136                                                                                                                                                                                    
        /dev/video10 support capture                                                                                                                                                                               
[NOTICE] src/V4l2Device.cpp:138                                                                                                                                                                                    
        /dev/video10 support read/write               
[NOTICE] src/V4l2Device.cpp:139
        /dev/video10 support streaming
[DEBUG] src/V4l2Device.cpp:71
        /dev/video10:YU12 size:1920x1080 bufferSize:3110400
[NOTICE] src/V4l2Device.cpp:225
        /dev/video10:YU12 size:1920x1080 bufferSize:3110400
[DEBUG] src/V4l2Device.cpp:71
        /dev/video10:YU12 size:1920x1080 bufferSize:3110400
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video10
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video10 nb buffer:8
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:0 size:3112960 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:1 size:3112960 offset:3112960
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:2 size:3112960 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:3 size:3112960 offset:3112960
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:4 size:3112960 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:5 size:3112960 offset:3112960
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:6 size:3112960 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video10 buffer idx:7 size:3112960 offset:3112960
[NOTICE] src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
[NOTICE] src/V4l2Device.cpp:135
        /dev/video11 support output
[NOTICE] src/V4l2Device.cpp:136
        /dev/video11 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video11 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video11 support streaming
[DEBUG] src/V4l2Device.cpp:71
        /dev/video11:BGR4 size:1920x1080 bufferSize:8294400
[ERROR] src/V4l2Device.cpp:212
        /dev/video11: Cannot set pixelformat to:h264 format is:BGR4
[ERROR] src/V4l2Device.cpp:83
        Cannot init device:/dev/video11
[NOTICE] src/V4l2MmapDevice.cpp:141
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
        Device /dev/video11
[WARN] src/v4l2compress.cpp:40
        Cannot create V4L2 output interface for device:/dev/video11
[NOTICE] src/V4l2MmapDevice.cpp:141

`

mpromonet, 2021-08-03:
Hi,

Did you try with `H264` instead of `h264` ?

Best Regards,
Michel.

owen7900, 2021-08-03:
H264 rather than h264 fixed the problem. Thank you

closed 2021-08-03 by owen7900