mpromonet/v4l2tools #46: v4l2compress: command not found

tobe2bcon, 2022-02-26:
Hi Michel, 
since I couldn't find any useful information, I'm afraid I need your help now.

**Background**
I have installed v4l2rtspserver on a Tinker Edge T board with Mendel OS (Debian derivate). The server is running fine and provides an UVC cam's stream as rtsp. I then installed v4l2tools via snapd as depicted in the v4l2rtspserver docs.
modprobe v4l2loopback video_nr=10 creates a virtual device successfully. I have proved this via v4l2-ctl --list-devices and got both the cam port (video2) and the virtual port (video10).

**Describe the bug**
Running
`v4l2compress -fH264 /dev/video0 /dev/video10 &`
results in this error:
`-bash: v4l2compress: command not found`

I also tried to activate the v4l2compress module:
`sudo modprobe v4l2compress`
and got:
`modprobe: FATAL: Module v4l2compress not found in directory /lib/modules/4.14.98-imx`

Do you have an idea what has to be done in this case? Thank you very much in advance

Best
Thomas

mpromonet, 2022-03-01:
Hi Thomas,

v4l2compress is an executable, not a kernel module.
If you installed it thought snap, you should be able to start it using `v4l2tools.v4l2compress`. 
Running `snap info v4l2tools` should helps you.

Best Regards,
Michel.

tobe2bcon, 2022-03-01:
> If you installed it thought snap, you should be able to start it using v4l2tools.v4l2compress ools should helps you.

Hi Michel,

thank you very much for your helpful explanation. I wasn‘t aware that the correct command is v4l2tools.v4l2compress. I just took v4l2compress☹️

Merci bien

Best
Thomas

closed 2022-03-07 by mpromonet