mpromonet/v4l2tools #47: Compress YUYV to h264 out

Tormund1985, 2022-05-18:
**Describe the bug**
I am trying to make a rtsp workflow from a libcamera sensor (IMX519 with autofocus).

**To Reproduce**
sudo modprobe v4l2loopback video_nr=3

v4l2compress -fH264 /dev/video0 /dev/video3

```
root@raspberrypi:/# v4l2compress -fH264 /dev/video0 /dev/video3
log level:500

[NOTICE] src/V4l2Device.cpp:133
        driver:unicam capabilities:a5a00001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video0:YUYV size:640x480 bufferSize:614400
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video0
[NOTICE] src/V4l2MmapDevice.cpp:73
VIDIOC_STREAMON: Invalid argument
        Device /dev/video0 nb buffer:10
[NOTICE] src/V4l2MmapDevice.cpp:141
        Device /dev/video0
[WARN] src/v4l2compress.cpp:215
        Cannot create V4L2 capture interface for device:/dev/video0
```

**Expected behavior**
Cannot create the v4l2 capture for proceed qith rtsp workflow.

**Desktop (please complete the following information):**
Raspberry Pi Zero 2w Bullseye

**Additional context**

I follow the steps to install v4l2tools, v4l2loopback and v4l2rtspserver without errors.

mpromonet, 2022-05-18:
Hi,

You may increase log level to get more information.

Best Regards,
Michel.

Tormund1985, 2022-05-19:
I am sorry, I am gonna try to supply more info. 

Journalctl give me next message:

```
may 19 08:04:53 raspberrypi kernel: unicam 3f801000.csi: Wrong width or height 640x480 (remote pad set to 4656x3496)
may 19 08:04:53 raspberrypi kernel: unicam 3f801000.csi: Failed to start media pipeline: -22
```

Any log in particular?

Many thanks for your support .

mpromonet, 2022-05-19:
Hi,

I was speaking about increasing log verbosity of the application v4l2compress, from the log you provide, we just see /dev/video0 cannot be openned.

Best Regards,
Michel.

Tormund1985, 2022-05-19:
> Hi,
> 
> I was speaking about increasing log verbosity of the application v4l2compress, from the log you provide, we just see /dev/video0 cannot be openned.
> 
> Best Regards, Michel.

I change resolution to 4656x3496 but still not working. With -vv command show:

```
root@raspberrypi:/# v4l2compress -vv -fH264 /dev/video0 /dev/video3
log level:700

[NOTICE] src/V4l2Device.cpp:133
        driver:unicam capabilities:a5a00001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video0 support streaming
[DEBUG] src/V4l2Device.cpp:71
        /dev/video0:YUYV size:4656x3496 bufferSize:32554752
[NOTICE] src/V4l2Device.cpp:225
        /dev/video0:YUYV size:4656x3496 bufferSize:32554752
[DEBUG] src/V4l2Device.cpp:71
        /dev/video0:YUYV size:4656x3496 bufferSize:32554752
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video0
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video0 nb buffer:7
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:0 size:32554752 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:1 size:32554752 offset:32555008
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:2 size:32554752 offset:65110016
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:3 size:32554752 offset:97665024
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:4 size:32554752 offset:130220032
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:5 size:32554752 offset:162775040
[INFO] src/V4l2MmapDevice.cpp:92
VIDIOC_STREAMON: Invalid argument
        Device /dev/video0 buffer idx:6 size:32554752 offset:195330048
[NOTICE] src/V4l2MmapDevice.cpp:141
        Device /dev/video0
[WARN] src/v4l2compress.cpp:215
        Cannot create V4L2 capture interface for device:/dev/video0root@raspberrypi:/#
```

mpromonet, 2022-05-19:
Hi,

It seems the req buffers to the v4l2 kernel module fails, maybe a bug there. You can try with '-r' to use the read interface instead of mmap.

Best Regards,
Michel.

Tormund1985, 2022-05-19:
Executing:
`modprobe v4l2loopback video_nr=3 
`
And then:
`root@raspberrypi:/# v4l2compress -r -fH264 /dev/video0 /dev/video3 & v4l2rtspserver -F 10 -S -P 8554 /dev/video3
[1] 804
log level:500

[NOTICE] src/V4l2Device.cpp:133
        driver:unicam capabilities:a5a00001 mandatory:1000001
[NOTICE] src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video0:pRAA size:1280x720 bufferSize:1152000
[NOTICE] src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
[NOTICE] src/V4l2Device.cpp:135
        /dev/video3 support output
[NOTICE] src/V4l2Device.cpp:136
        /dev/video3 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video3 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video3 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video3:H264 size:1280x720 bufferSize:3686400
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video3
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video3 nb buffer:2
[NOTICE] src/v4l2compress.cpp:74
log level:500

[NOTICE] /home/pi/v4l2rtspserver/main.cpp:297
        Version: 0.3.3-2-g0c63d58 live555 version:2022.04.26
[NOTICE] /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
        Create V4L2 Source.../dev/video3
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85208003 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:135
        /dev/video3 support output
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
        /dev/video3 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
        /dev/video3 support read/write
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
        /dev/video3 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
        /dev/video3:H264 size:1280x720 bufferSize:3686400
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:246
        fps:1/10
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:247
        nbBuffer:2
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
        Device /dev/video3
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
        Device /dev/video3 nb buffer:2
[NOTICE] /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
        Create Source .../dev/video3
[NOTICE] /home/pi/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
        begin thread
[NOTICE] /home/pi/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
        format:video/H264
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:80
        Play this stream using the URL "rtsp://192.168.31.87:8554/ts"
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:163
        HLS       http://192.168.31.87:8554/ts.m3u8
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:164
        MPEG-DASH http://192.168.31.87:8554/ts.mpd
[NOTICE] /home/pi/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
        format:video/H264
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:80`

getting this error throught journal:

```
may 19 12:30:34 raspberrypi kernel: unicam 3f801000.csi: Wrong width or height 640x480 (remote pad set to 4656x3496)
may 19 12:30:34 raspberrypi kernel: unicam 3f801000.csi: Failed to start media pipeline: -22
```

It works but video not showing on rtsp streaming.

Of course camera with libcamera commands works fine.

mpromonet, 2022-05-20:
Hi,

v4l2compress doesnot change the resolution, it takes capture size and compress with same size. You may set resolution with v4l2-ctl before running v4l2compress.

Best Regards,
Michel.

Tormund1985, 2022-05-25:
HI Michel,

Nothing change for me:

```
root@raspberrypi:/# v4l2compress -v -r -fH264 /dev/video0 /dev/video3
log level:600

[NOTICE] src/V4l2Device.cpp:133
        driver:unicam capabilities:a5a00001 mandatory:1000001
[NOTICE] src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video0:pRAA size:1920x1080 bufferSize:2592000
[NOTICE] src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
[NOTICE] src/V4l2Device.cpp:135
        /dev/video3 support output
[NOTICE] src/V4l2Device.cpp:136
        /dev/video3 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video3 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video3 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video3:H264 size:1920x1080 bufferSize:8294400
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video3
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video3 nb buffer:2
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video3 buffer idx:0 size:8294400 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video3 buffer idx:1 size:8294400 offset:8294400
[NOTICE] src/v4l2compress.cpp:74
**Violación de segmento**
```

```
root@raspberrypi:/# v4l2compress -v -fH264 /dev/video0 /dev/video3
log level:600

[NOTICE] src/V4l2Device.cpp:133
        driver:unicam capabilities:a5a00001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video0:pRAA size:1920x1080 bufferSize:2592000
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video0
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video0 nb buffer:10
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:0 size:2592000 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:1 size:2592000 offset:2592768
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:2 size:2592000 offset:5185536
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:3 size:2592000 offset:7778304
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:4 size:2592000 offset:10371072
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:5 size:2592000 offset:12963840
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:6 size:2592000 offset:15556608
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:7 size:2592000 offset:18149376
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:8 size:2592000 offset:20742144
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video0 buffer idx:9 size:2592000 offset:23334912
[NOTICE] src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85208003 mandatory:4000002
[NOTICE] src/V4l2Device.cpp:135
        /dev/video3 support output
[NOTICE] src/V4l2Device.cpp:136
        /dev/video3 support capture
[NOTICE] src/V4l2Device.cpp:138
        /dev/video3 support read/write
[NOTICE] src/V4l2Device.cpp:139
        /dev/video3 support streaming
[NOTICE] src/V4l2Device.cpp:225
        /dev/video3:H264 size:1920x1080 bufferSize:8294400
[NOTICE] src/V4l2MmapDevice.cpp:49
        Device /dev/video3
[NOTICE] src/V4l2MmapDevice.cpp:73
        Device /dev/video3 nb buffer:2
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video3 buffer idx:0 size:8294400 offset:0
[INFO] src/V4l2MmapDevice.cpp:92
        Device /dev/video3 buffer idx:1 size:8294400 offset:8294400
[NOTICE] src/v4l2compress.cpp:74
V**iolación de segmento**
```

Any ideas?

Thanks you so much for your help, I really appreciate it.