mpromonet/v4l2tools #49: Cant install v4l2tools

drincon7, 2023-03-18:
Im trying to use V4l2compress to convert an video capture with UYVY format to H264. The problem is i cant install V4l2tools.
When i make: $make i receive this:
```
PREFIX=/usr
ARCH=x86_64
with log4cpp
with x264
with x265
with vpx
with jpeg
git submodule update --init v4l2wrapper
error: pathspec 'v4l2wrapper' did not match any file(s) known to git
make: *** [Makefile:113: v4l2wrapper/Makefile] Error 1 
```

I am on Ubuntu 22.04.1 LTS
thanks for your answer in advance.

mpromonet, 2023-03-18:
Hi, 

Did you clone the repo v4l2tools ?

Best Regards,
Michel.

drincon7, 2023-03-18:
Hello, yes i did it, and it not work..

mpromonet, 2023-03-18:
Hi @drincon7 

Your problem is probably related to git, the command that fails seems `git submodule update --init v4l2wrapper`, you can try to run it, adding some log level to get more information.

Best Regards,
Michel.