mpromonet/v4l2tools #51: Make fails with error "Submodule path 'h264bitstream': checked out"

rsfitz, 2023-07-09:
**Description**
Build currently fails to complete; returning the following error:

```
g++ -o v4l2copy -std=c++11 -W -Wall -pthread -g -pipe  -I include -I v4l2wrapper/inc src/v4l2copy.cpp libv4l2wrapper.a 
g++ -o v4l2source_yuv -std=c++11 -W -Wall -pthread -g -pipe  -I include -I v4l2wrapper/inc src/v4l2source_yuv.cpp libv4l2wrapper.a 
git submodule update --init h264bitstream	
Submodule 'h264bitstream' (https://github.com/aizvorski/h264bitstream) registered for path 'h264bitstream'
Cloning into '/home/v4l2tools/h264bitstream'...
Submodule path 'h264bitstream': checked out '34f3c58afa3c47b6cf0a49308a68cbf89c5e0bff'
cd h264bitstream && autoreconf -i -f && ./configure --host arm-linux-gnueabihf
/bin/sh: 1: autoreconf: not found
make: *** [Makefile:142: h264bitstream/.libs/libh264bitstream.so] Error 127 
```

**To Reproduce**
1. Clone Repository
2. Run Make

mpromonet, 2023-07-09:
Hi

Error message says autoreconf is not found, you should install it.

Best Regards
Michel

rsfitz, 2023-07-09:
Apologies, should have caught that.  Assumed it was a issue with linked submodule.

closed 2023-07-09 by rsfitz